MickiusMousius/Rolio46Keyboard #9: Version 1.0 Purchase Requests

dickiedyce, 2023-10-11:
Is there somewhere we can note our interest in purchasing v1.0 ?

MickiusMousius, 2023-10-11:
Not yet, I'll set something up tonight using Google forms though.

As a point of curiosity would you be wanting a fully built unit or a kit? Also, would you want the aluminium chassis or a plastic one?

MickiusMousius, 2023-10-13:
Hey, I finally got the survey done, I needed too set up a company for a few bits first.

Link below: https://forms.gle/bdBZMKoUopvWX2VN7

If this works for you I'll post it more visibly soon.

closed 2023-10-14 by MickiusMousius

dickiedyce, 2023-10-18:
I have used the form - really great. Put me down for 4 !

dickiedyce, 2023-10-18:
I've just seen it on KBD News - https://kbd.news/Rolio-0.9-2158.html

MickiusMousius, 2023-10-19:
Ha, that's cool.

There's even a picture of the prototype travel case! I maybe should have re-printed the case with the fixed up logo.

I can see you've answered the survey, thank you. If you do genuinely want one of these boards (though you do say 4) I can probably get you at least one soonish, I have 6 aluminium chassis coming from JLCPCB soon. I could sell you one as an early test, if you really like it I can do more after, with some potential for limited customisation if you'd like (LED and chassis colour for example).

Let me know here if you're keen and I'll email you directly with more information.

dickiedyce, 2023-10-19:
That would be FANTASTIC!
I say 4, but I don't want to appear greedy ;-) . I have a Corne-ish Zen, which I love, but I only managed to get one, and it's awkward for work & home. If it's a limited run, I completely understand that from your point of view, more users is better...
But .... your extra keys and scrollers just looks cracking.