MickiusMousius/Rolio46Keyboard #13: Any more information on the vista display modules?

milch, 2024-01-23:
Hey,

The nice!view seems to be out of stock everywhere at the moment so I've been looking at alternatives and found this repo. I unfortunately can't seem to find any more information on the Vista modules you describe via google other than a reference back to this repo, do you have any more information about it like a seller or manufacturer? 

Thank you!

MickiusMousius, 2024-01-24:
Hi,

I have about 50 Vista272 modules in stock (nice view “clone”) and 150 vista508 modules.

If you’re in need I can sell you some, just reach out to contact@keydio.io and we’ll try to help.

Cheers

Ryan

joric, 2024-04-07:
@MickiusMousius are they open source? Who makes/sells them? Karnadii module doesn't seem to be size-compatible with nice!view https://github.com/joric/nrfmicro/wiki/ZMK#lcd

MickiusMousius, 2024-04-08:
Hi, they’re not open source. Though I likely will open them up soon enough. The design is pretty simple, the only value in it is the part numbers and the time taken to assemble them.

I make them, I’ve been intending to sell them via keydio.io. I have plenty of stock, but have yet to finish setting the store up as I’ve been spending a lot of time visiting a sick family member instead of working on the store.