MickiusMousius/Rolio46Keyboard #14: [PCB] EVQWGD001 seems to be routed/labeled incorrectly

mbiernacki89, 2024-09-20:
Hi, 
Since the store seems to be delayed, I decided to build Rolio on my own.
When looking at the PCB project, I've noticed that the EVQWGD001 encoders seem to be routed incorrectly.

Comparing your schematics to the PCB: 
[image][image]

Pins A and B are located in the middle of the 4-pin section, while routing on your PCB points to pin 1 and 2 (looking from the switch pins side),

Am I missing something here? (Mind you I'm a total novice when it comes to building PCBs)

MickiusMousius, 2024-09-20:
I may have something out of sync there in the repo. I’ll try and have a look this week, I’m out of town at the moment which is why I can’t look right now.

If it helps at all the only delay with the store has been getting the shopfront up. I have plenty of inventory so can ship populated and partially populated PCBs up to a fully assembled keyboard.

If you’d like more info just reach out via contact@keydio.io 😊

Cheers

Ryan

mbiernacki89, 2024-09-20:
Thanks for a fast response! :)
I've soldered the pcb In the meantime and the encoder seems to work with the routing you have there so it might have been a false alarm on my part, sorry for that.

But since we are in topic of things out of sync. It seems that `JLCPCB - Gerbers - Right` zip file actually contains files for left pcb instead. I zipped the Gerbers manually before ordering but it could be overlooked by someone

MickiusMousius, 2024-09-20:
Thanks for that, per comment left on the other issue I’ll make an effort to update the repo with all of the latest models and files.

I suspect that this might just be a footprint vs. schematic picture thing where the order of the pins in the electrical schematic may not match the  order of the pins on the physical footprint. Though  without KiCAD at hand that’s just a guess.