MickiusMousius/Rolio46Keyboard #15: SLS chassis - right bottom plate don't have enough space for the R3 680Ω resistor

mbiernacki89, 2024-09-20:
The right bottom plate (chassis/sls/fdm pla parts/right - bottom plate.stl) seems to not have enough space for one of the 680Ω resistors (marked R3). I had to carve a portion of it to make space for it.

[image]

MickiusMousius, 2024-09-20:
Ah, I’ve fixed that a long time ago.

I’ve got a lot of updated CAD models, lots of improvements have been made to the chassis and travel cases.

I think I will need to prioritise updating these repos with the newest models.

mbiernacki89, 2024-09-20:
Ah, I'm glad I havent ordered cnc chassis right away then :)

I'll be waiting for the updates, thanks for the help and quick responses

closed 2024-09-20 by mbiernacki89