MickiusMousius/Rolio46Keyboard #16: Is it open source?

WMnologo, 2024-09-23:
[image]
Rolio46 is so cool, is this CNC file open source?

WMnologo, 2024-09-23:
[image]
I searched and couldn't find one of the files. Is it open source

MickiusMousius, 2024-09-24:
Ah, that one is not yet open source yet. But I’ll publish it soon. I have one more iteration to make it easier to assemble and improve Bluetooth performance.

I do have some inventory on hand though.

WMnologo, 2024-09-27:
﻿
Okay, looking forward to your open source. It's really cool and I really like it.

WMnologo, 2024-09-27:
[image]
Can you share this  model? I want the keyboard to be easy to carry around

MickiusMousius, 2024-09-30:
I shall, if you’re building one now let me know, otherwise I’ll try to batch it together with a bunch of other updates.

WMnologo, 2024-10-05:
Yes, I made one. Now I want this shell to be easy to carry around

MickiusMousius, 2024-10-05:
Did you make a CNC chassis or a 3D printed one?

the two travel cases have different internal dimensions.

WMnologo, 2024-10-07:
3D printed

WMnologo, 2024-10-15:
hello

MickiusMousius, 2024-10-18:
Hey, I think I have the file. But can not attach a step file here it seems.

I don’t want to put it in the repo as I have a heavily revised FDM chassis (no glue needed now) and it requires a different case.

How would you like for me to share the file to you?

WMnologo, 2024-11-14:
Have his 3D files and PCB files been open source? I try one. It's so cool.
[image]